adafruit/Adafruit_CircuitPlayground #11: API Documentation

makenova, 2016-09-03:
Hi,
  This is a great library but I struggled with using it because I didn't know where to look for the the available methods. I just happened to stumble upon some of the methods that I needed in tutorials and then I found the rest in the header files.
I think the addition of a `README.md` file would be beneficial to users. 

Something along these lines:

```md
# Adafruit_CircuitPlayground

## Methods

### boolean CircuitPlayground.begin(uint8_t brightness)

Initializes the library.

...

### boolean leftButton(void)

check the state of the left button.

### boolean rightButton(void)

check the state of the right button.
...

### void CircuitPlayground.senseColor(uint8_t& red, uint8_t& green, uint8_t& blue)

Perform basic RGB color sensing.
The return values are assigned to the parameters.

### uint32_t CircuitPlayground.senseColor()

Perform basic RGB color sensing.

...
```

I was eventually able to do what I wanted, so I do not thing this is required, but I do think it will
lower the barrier to entry.

Thank you!

ladyada, 2016-09-03:
we have a link here
https://learn.adafruit.com/circuit-playground-lesson-number-0/library-reference
more to come! :)

closed 2016-09-03 by ladyada

makenova, 2016-09-03:
@ladyada Thanks!

If anyone else stumbles on this issue, follows that link, and is confused by the blank page, there's another link to:

https://caternuson.github.io/Adafruit_CircuitPlayground/