adafruit/Adafruit_CircuitPlayground #14: 'analogPinToChannel' not declared in scope : Adafruit_CPlay_Mic.cpp

monsieurDavid, 2016-10-17:
- Arduino board:  Circuit PlayGround Dev. Edition

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.12

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
Compile and upload `~/Arduino/libraries/Adafruit_Circuit_Playground/examples/demo.ino`

```
~/Arduino/libraries/Adafruit_Circuit_Playground/utility/Adafruit_CPlay_Mic.cpp: In member function 'int Adafruit_CPlay_Mic::peak(uint16_t)':
~/Arduino/libraries/Adafruit_Circuit_Playground/utility/Adafruit_CPlay_Mic.cpp:27:37: error: 'analogPinToChannel' was not declared in this scope
   channel     = analogPinToChannel(4); // Pin A4 to ADC channel
                                     ^
~/Arduino/libraries/Adafruit_Circuit_Playground/utility/Adafruit_CPlay_Mic.cpp: In member function 'void Adafruit_CPlay_Mic::capture(int16_t*, uint8_t)':
~/Arduino/libraries/Adafruit_Circuit_Playground/utility/Adafruit_CPlay_Mic.cpp:84:37: error: 'analogPinToChannel' was not declared in this scope
   channel     = analogPinToChannel(4); // Pin A4 to ADC channel
                                     ^
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

monsieurDavid, 2016-10-17:
My mistake - was uploading to wrong board. Nothing to see. Move along.

closed 2016-10-17 by monsieurDavid

sigmus731, 2017-02-01:
What exactly do you mean by saying the wrong board?
Sorry, but i'm just starting to learn about arduino programming.

jtrombone, 2018-01-23:
I found that on my Bare-Bones board (I use the Arduino/Genuino Uno board option I just had to comment out the line:
//#include <Adafruit_Circuit_Playground.h>
I believe this line is for the specific Adafruit board
Also had to add the average.h header file in the sketch folder
Found the header file here: 
https://github.com/MajenkoLibraries/Average/blob/master/Average.h