adafruit/Adafruit_CircuitPlayground #17: Should playTone() be blocking?

caternuson, 2016-11-02:
- Arduino board:  Circuit Playground  (library up to 1.6.3)
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.6.11
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): Use playTone() without delay()

Ref. this forum thread:
https://forums.adafruit.com/viewtopic.php?f=58&t=102459

Ref. current library documentation:
https://learn.adafruit.com/circuit-playground-lesson-number-0/mini-speaker#library-reference

Ref. CircuitPlayground library playTone() function:
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/master/Adafruit_CircuitPlayground.cpp#L79

Ref. Arduino library tone() function:
https://www.arduino.cc/en/Reference/Tone

The current Circuit Playground `playTone()` function simply calls the Arduino library `tone()` function. The tone generation is done using timer interrupts. The call to `tone()` configures the timer and returns. This leads to the following lines of code only producing the second (800Hz) sound:

```C++
CircuitPlayground.playTone(500, 1000);
CircuitPlayground.playTone(800, 1000);
```

This is fixed quite simply using delay():

```C++
CircuitPlayground.playTone(500, 1000); delay(1000);
CircuitPlayground.playTone(800, 1000); delay(1000);
```

However, a blocking behaviour may be better suited for CircuitPlayground. Or at least the option for both (blocking/non-blocking) with blocking being the default.

ladyada, 2016-11-02:
oh thats odd, i thought tone() was blocking? ok well which do you think we should do? :)

caternuson, 2016-11-02:
Guess it's a point of discussion. I'd go with having both available, but with the default being blocking. I think that behaviour is easier to understand for the beginner. However, not sure how to implement that at this point without breaking backwards compatibility.

In terms of implementation, I see two options:

1. Prototype `playTone()` with a boolean third parameter to control blocking and give it a default value (which would be non-blocking to maintain backwards compatibility).
2. Create a new function that is blocking. `playTone2()`, `playToneAndWait()`, `playToneLōc()` ?

ladyada, 2016-11-02:
yah - i think blocking by default is wise, do you mind doing a pull req with the change? ill merge it in - just looking at another issue right now :)

caternuson, 2016-11-02:
So...OK to break backwards compatibility? i.e., change to:
```C++
playTone(800, 1000);        // blocking call
playTone(800, 1000, true);  // same as above
playTone(800, 1000, false); // non-blocking call
```
Yes. Can implement and make PR.

ladyada, 2016-11-02:
yah im ok with breaking

ladyada, 2016-11-07:
resolved!

closed 2016-11-07 by ladyada