adafruit/Adafruit_CircuitPlayground #25: Examples do not compile

gregberger, 2017-10-19:
## Setup
- Arduino board:  Circuit Playground Express ATSAMD21
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.3 (MacOS and Ubuntu 17.04)

## Reproduction of the bug
Installed Arduino's SAMD Boards via the Board manager (I had no luck at all to get CPE working with the Adafruit's AVR Boards manager)
Installed the last version of Circuit Playground Library (1.6.9) via the Library manager
Run the demo example.

The example won't compile and I Got an error message saying two things (macOs): 

1. `/Applications/Arduino/libraries/Adafruit_CircuitPlayground/utility/CP_Boards.h:708:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"`

2. That two libraries were found and only one was used. (I read this [thread](https://github.com/arduino/Arduino/issues/6462) which clarifies a bit the situation)

## Temporary fix
After multiple reinstallation of the board manager and the library, I finally got the examples to compile (with a lot of warnings!) and upload on board by changing the `#include` directive from `<Adafruit_CircuitPlayground.h>` to `<Adafruit_Circuit_Playground.h>` (notice the third underscore).

### Suggestion
I am a bit puzzled by that behaviour and wanted to know if I am alone. 
If not, maybe we could fix that by changing the `#include` in all examples ?

ladyada, 2017-10-19:
the latest IDE is required (we recently fixed this bug) can you update to 1.8.5?

gregberger, 2017-10-19:
Yes! updating to 1.8.5 did the trick. 

I wasn't aware my version of the IDE was two patches behind, sorry for wasting your time

Thanks!

closed 2017-10-19 by gregberger

ladyada, 2017-10-20:
all good! thanks for mentioning it - we'll warn people in the guide :D