adafruit/Adafruit_CircuitPlayground #30: soundPressureLevel() intermittently returns 'inf' at low levels on AVR

PaintYourDragon, 2018-01-05:
In the dm-mic-spl fork, running the soundPressureLevel example. Open serial monitor (not the plotter) and (when quiet) you'll see some "inf"s mixed in with the numeric output. Only happens on AVR; ARM seems unaffected. @dhalbert

ladyada, 2018-01-05:
oops thats for @deanm1278 :)

PaintYourDragon, 2018-01-05:
Wrong button. Sorry! :)

deanm1278, 2018-01-05:
oh oops I had fixed this yesterday but must have messed up with git. It should have been doing it with both AVR and ARM, just ARM is a bit lower of a floor.
should be fixed here: https://github.com/adafruit/Adafruit_CircuitPlayground/commit/69e723712ff05f2237f4e0e3e5a212f28e03c163

It's a bit too loud to test right now at the factory but I'll plug it in when I get home.

ladyada, 2018-01-21:
i think this was resolved so closin!

closed 2018-01-21 by ladyada