adafruit/Adafruit_CircuitPlayground #33: Having trouble with the Speech examples

maxicusj, 2018-03-10:
I am having trouble with the Speech examples… Getting just buzz and hiss, no speech. Arduino IDE 1.8.5, Playground Circuit 1.8.0 library. Chip is Playground Circuit classic. Does anyone have a clue?
Rest of the examples seem to work… Tried on Windows and OSX and in Arduino Web Editor  – same story, no speech, just hissing and buzzing.

Please, suggest?

Thanks,
Roman

ladyada, 2018-03-14:
hmm can you try the Talkie library direct? you'll  need to install it and run the demos but it  *ought* to work?
https://github.com/adafruit/Talkie

manitou48, 2018-03-15:
On playground classic (AVR mega32u4),  circuit playground speech examples are just noise with Arduino 1.8.4 and 1.8.5, with the latest github version 1.8.  I reverted to Arduino  1.8.1 and examples worked.

maxicusj, 2018-03-15:
Thank you, guys! with the Arduino version 1.8.0 all working fine. Talkie too.

manitou48, 2018-03-18:
I thought the problem was with the compilers for Arduino IDE 1.8.4 and 1.8.5, but further testing shows that Arduino 1.8.4 speech **examples work if I use version 1.6.4 of CircuitPlayground library.**  The speech is noise with library version 1.6.8 and 1.8.0.  Arduino 1.8.4 ships with lib 1.6.8

maxicusj, 2018-03-18:
strange, I got Arduino 1.8.1 and CircuitPlayground library 1.8.0 and that seems to work OK. Development platform is OSX.

manitou48, 2018-03-18:
Nope, not for me with Ubuntu.  I had to remove the arduino 1.8.1 version of circuitplayground library for sketch to use my sketchbook/llibrary version (1.8.0).  speech examples are noise with arduino 1.8.1 and lib 1.8.0.   (are you sure IDE is using lib 1.8.0 and not builtin 1.6.4?)

maxicusj, 2018-03-18:
no, not sure. But 1.8.0 is what I see in the Library Manager as being installed. How else can one check?

manitou48, 2018-03-18:
you'll need to remove/mv the IDE version of the CircuitPlayground library (and restart IDE).  you might look around to see if other copies are lurking.

maxicusj, 2018-03-18:
ok, I think I will leave it the way it is. Don't touch the running system :-) thanks for tips and solving this.

manitou48, 2018-03-19:
ladyada, I think the problem with broken speech examples in 1.8.0 lib  is in utility/Adafruit_CPlay_Speaker.cpp in the begin() method.  lib 1.6.4
 has
`void Adafruit_CPlay_Speaker::begin(void) {
  // Set up Timer4 for fast PWM on !OC4A
  PLLFRQ  = (PLLFRQ & 0xCF) | 0x30;   // Route PLL to async clk
  TCCR4A  = _BV(COM4A0) | _BV(PWM4A); // Clear on match, PWMA on
  TCCR4B  = _BV(PWM4X)  |_BV(CS40);   // PWM invert, 1:1 prescale
  TCCR4D  = 0;                        // Fast PWM mode
  TCCR4E  = 0;                        // Not enhanced mode
  TC4H    = 0;                        // Not 10-bit mode
  DT4     = 0;                        // No dead time
  OCR4C   = 255;                      // TOP
  OCR4A   = 127;                      // 50% duty (idle position) to start
  started = true;
  pinMode(5, OUTPUT);                // Enable output
}`
 but lib 1.8.0 only has pinMode().  Adding the  additional code to 1.8.0 begin allowed speech examples to work.

ladyada, 2018-03-19:
could be!! @PaintYourDragon wanna take a look?

manitou48, 2018-03-19:
It may take a little more work and regression testing, because i noticed that the beep/tones of demo example no longer work with  Speaker::begin() mods above.   All work with lib 1.6.4

manitou48, 2018-03-21:
Latest attempt at patching lib 1.8.0.  Rather than pasting PLLFRQ stuff in CPlay_Speaker::begin(),  I added it into say() in talkie.cpp as below
`    if(!started) {
      begin();
#ifdef __AVR__
    // Set up Timer4 for fast PWM on !OC4A
    PLLFRQ  = (PLLFRQ & 0xCF) | 0x30;   // Route PLL to async clk
    TCCR4A  = _BV(COM4A0) | _BV(PWM4A); // Clear on match, PWMA on
    TCCR4B  = _BV(PWM4X)  |_BV(CS40);   // PWM invert, 1:1 prescale
    TCCR4D  = 0;                        // Fast PWM mode
    TCCR4E  = 0;                        // Not enhanced mode
    TC4H    = 0;                        // Not 10-bit mode
    DT4     = 0;                        // No dead time
    OCR4C   = 255;                      // TOP
    OCR4A   = 127;                      // 50% duty (idle position) to start
    started = true;
#endif
    }
`

speech examples play and tone/beep still works in demo

closed 2019-01-25 by PaintYourDragon