adafruit/Adafruit_CircuitPlayground #36: Adafruit_SleepyDog.h not found when compiling megademo example.

ReanimationXP, 2018-07-21:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **Adafruit Circuit Playground Classic**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.6** (Hourly)

When flashing the Circuit Playground library's [megademo](install:
   - arduino --install-library "Adafruit LED Backpack Library","Adafruit GFX Library","Adafruit SleepyDog Library","Adafruit Zero FFT Library") example, I was presented with an error citing that `Adafruit_SleepyDog.h` was missing. After searching the github I only see it referenced one other place, which is [.travis.yml](https://github.com/adafruit/Adafruit_CircuitPlayground/blob/abc0cad00459656f402945df6b89287a7d678297/.travis.yml). Specifically:

```
install:
   - arduino --install-library "Adafruit LED Backpack Library","Adafruit GFX Library","Adafruit SleepyDog Library","Adafruit Zero FFT Library"
```

I checked my libraries and found that this was not installed, despite having just recently installed the Adafruit Circuit Playground library.  I tried deleting the Playground library and then reinstalling it, which worked fine, but SleepyDog still was not present. Only after searching for it manually in the library manager and installing it did megademo compile and flash correctly.

I'd venture to guess something is going wrong with the install script. I wasn't aware libraries could even install other libraries as dependencies. I assume that's what's supposed to be happening here and that'd be a neat and interesting feature if it worked :) Ideas?

My hourly is a bit old but it's still newer than the latest release:
`1.8.6 Hourly Build 2018/04/19 10:37`

ReanimationXP, 2018-07-21:
Also worth noting that other libraries here like LED Backpack Library were also not installed.

ladyada, 2018-07-22:
hi it's mentioned here! :)
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/cc82c9165066b80afb715eb46153a85857fd5f20/examples/mega_demo/mega_demo.ino#L13

closed 2018-07-22 by ladyada

ReanimationXP, 2018-07-22:
Okay, but why is it mentioned in the `-install-library` flag then?

ladyada, 2018-07-22:
thats so travis can compile things, you can read about it here
https://learn.adafruit.com/the-well-automated-arduino-library

ReanimationXP, 2018-07-22:
Ah, thanks. Found it after posting. Didn't occur to me it'd have nothing to do with Arduino. Thanks :)