adafruit/Adafruit_CircuitPlayground #40: redLed() no longer working for CPC

caternuson, 2019-05-28:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=58&t=152174

Seems to be related to low level register writes and timer setup in `Adafruit_CPlay_Speaker::begin()`: 
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/991fd39f8860a7fed0e08b7fb6d69192b022c783/utility/Adafruit_CPlay_Speaker.cpp#L18-L29

ladyada, 2019-05-28:
if you remove that section, does the LED start working? :/

caternuson, 2019-05-28:
Yep.

Looks like OC4A is on D13:
[image]

ladyada, 2019-05-28:
dya mind checking the DS to see if we have a way to disable PWM output on OC4A

caternuson, 2019-05-28:
Not seeing an option to have only the complementary output (SPEAKER) connected. Only option to disconnect OC4A is to disconnect both:
[image]

That init section was moved from another location with this commit:
https://github.com/adafruit/Adafruit_CircuitPlayground/commit/f3694f46be6626ccdacad050bb598770c99b8f36

Maybe there's no need for the call to `speaker.begin()` call the c-tor anymore? `playSound()` will call it?

ladyada, 2019-05-28:
yep remove it - CPC's are not very popular anymore

caternuson, 2019-05-29:
Fixed by #41

closed 2019-05-29 by caternuson