adafruit/Adafruit_CircuitPlayground #42: Speaker on/off broken on CPX

caternuson, 2019-07-19:
Just need to configure digital 11 for output somewhere. The removal of the call to `speaker.begin()` in #41 removed that part of the setup.

Discovered while helping here:
https://forums.adafruit.com/viewtopic.php?f=58&t=154242

```cpp
#include <Adafruit_CircuitPlayground.h>

void setup() {
  CircuitPlayground.begin();
  //uncomment this and it will work again
  //pinMode(11, OUTPUT);
  CircuitPlayground.speaker.off();
}

void loop() {
  CircuitPlayground.playTone(500, 100);
  delay(1000);
}
```

ladyada, 2019-07-19:
hmm can you try that line to the begin() function :)

caternuson, 2019-07-19:
`Adafruit_CircuitPlayground::begin()`? That'd work. Basically need to re-engage this setup for CPX:
https://github.com/adafruit/Adafruit_CircuitPlayground/blob/412d9775aea73af8c1924f0c61a7d532b058a2fd/utility/Adafruit_CPlay_Speaker.cpp#L32-L37 
which isn't happening now since the call to `speaker.begin()` was removed to fix red LED on CPC.

ladyada, 2019-07-19:
? you can just set the pin to be an output without engaging the analogWrite?

caternuson, 2019-07-20:
Fixed with #43

closed 2019-07-20 by caternuson