adafruit/Adafruit_CircuitPlayground #46: Arduino Updater is always showing 1.86 New Version available

electronic-cosmos, 2019-11-01:
Header say everything :-)
Something is wrong in this version history.
Pleas check it, THX!

ladyada, 2019-11-01:
please post a screenshot

MartinL1, 2019-11-01:
@ladyada It's happening with my Arduino IDE Library Manager as well:

[image]

ladyada, 2019-11-01:
is this new or has always been happening?

MartinL1, 2019-11-01:
I'd say about a couple of weeks or so.

ladyada, 2019-11-01:
that is quite odd because the version number matches the release ... only thing that is new in the last few weeks is we went from 1.9.1 to 1.10.1 - i wonder if the parsing code for semver doesnt handle 2-digit minor numbers (kinda doubt thats it...)

@cmaglie do you have any ideas?

MartinL1, 2019-11-01:
I think I know what's happening. My Arduino IDE's got two copies of the Adafruit Circuit Python library. 

One, version 1.6.7 (judging by its library.properties file) is located at:

C:\Users\Computer\Documents\Arduino\libraries\Adafruit_Circuit_Playground

The other, version 1.8.6 is in the directory:

C:\Program Files (x86)\Arduino\libraries\Adafruit_Circuit_Playground

Removing the 1.6.7 version allows the Arduino IDE to correctly update to version 1.10.1.

MartinL1, 2019-11-01:
The updated 1.10.1 library is then correctly created in the directory:

C:\Users\Computer\Documents\Arduino\libraries\Adafruit_Circuit_Playground

electronic-cosmos, 2019-11-01:
For Mac users? What do  I have to do???

MartinL1, 2019-11-01:
I'm not a Mac user, but it might be located in the "libraries" directory where your Arduino sketches are located. In the "libraries" directory there should be "Adafruit_Circuit_Python" folder.

Failing that, it should be possible to search for the folder, to find its location.

electronic-cosmos, 2019-11-01:
I found it. 
1. Enter the Arduino package on mac Program folder.
2. go to: content/Java/libraries/Adafruit_Circuit_Playground
3. removing it.
4. also removing it in documents/Arduino/libraries/Adafruit_Circuit_Playground
5. Starting Arduino IDE again than like mentioned 1.10.1 is installed :-)
THX

electronic-cosmos, 2019-11-01:
Ups after updating the Adafruit_Circuit_Playground in the libraries manager.
Greetings from Reutlingen Germany
Michael J.

closed 2019-11-01 by electronic-cosmos