adafruit/Adafruit_CircuitPlayground #49: Capacitive touch not working on Bluefruit Playground

amiika, 2020-01-30:
It seems that capacitive touch is now working on CPB. For example this doesnt read anything:

```
CircuitPlayground.readCap(0)
```
Pinning seems to be different because some caps work, for example A6 = 0, A7 = 1 ... etc.

ladyada, 2020-01-30:
please update all board support packages & libraries and compile/run the 'demo' example - paste the output here :)

amiika, 2020-01-30:
Thanks! I see my issue now. I thought that those caps were from 0-7 and tried something like:

```
void loop() {
  for(int i=0;i<7;i++) {
    float val = CircuitPlayground.readCap(i);
    if(val>250) {
      Serial.print("Touched: "); Serial.println(i);
      delay(100);
    }
  }
 }
```

Didn't realize the cap numbers were so odd, but looked at the demo got it working:

```
void loop() {
  CircuitPlayground.speaker.enable(false);
  int caps[] = {6,9,10,3,2,0,1};
  for(int i=0;i<7;i++) {
    float val = CircuitPlayground.readCap(caps[i]);
    if(val>500) {
      Serial.print("Touched: A"); Serial.print(i+1); Serial.print("  Val: "); Serial.println(val);
      delay(100);
    }
  }
 }
```

Thanks again. You are the best! :)

closed 2020-01-30 by amiika