adafruit/Adafruit_CircuitPlayground #55: CircuitPlayground.readCap() breaks analogRead

DeqingSun, 2020-07-31:
```
#include <Adafruit_CircuitPlayground.h>

void setup() {
  CircuitPlayground.begin();
  Serial.begin(9600);
}

void loop() {

  int sensorValue = analogRead(A1);
  //CircuitPlayground.readCap(A2);

  Serial.println(sensorValue);

  delay(20);
}
```
This code works until ```CircuitPlayground.readCap(A2);``` is uncommented. Once I do that, analogRead(A1) can never reach 0. Even A1 is tied to GND.  It seems capacitive touch used QTOUCH lib so I'm not able to fix it myself this time.

DeqingSun, 2020-07-31:
After some poking around it seems 
```QTOUCH_PTC->BURSTMODE.reg = 0xA4;``` is the cause of trouble.

So after change measureRaw to this, it seems OK.

```
uint16_t Adafruit_CPlay_FreeTouch::measureRaw(void) {
  if (yline == -1) 
    return -1;

  runInStandby(true);
  enablePTC(true);
  // check if in progress
  // set up NVIC if using INT (we're not)
  enableWCOint(false);
  clearWCOintFlag();
  clearEOCintFlag();
  // enable_eoc_int(); // not using irq (for now)

  // set up pin!
  //Serial.print("Y Line #"); Serial.println(yline);
  selectYLine();
  // set up sense resistor
  setSeriesResistor(seriesres);
  // set up prescalar
  setOversampling(oversample);
  // set up freq hopping
  setFreqHopping(freqhop, hops);
  // set up compensation cap + int (?) cap
  setCompCap(compcap);
  setIntCap(intcap);
    uint32_t BURSTMODE_REG_BACKUP = QTOUCH_PTC->BURSTMODE.reg;

  QTOUCH_PTC->BURSTMODE.reg = 0xA4;

  sync_config();

    uint16_t result = startPtcAcquire();
    
    QTOUCH_PTC->BURSTMODE.reg = BURSTMODE_REG_BACKUP;
    
    sync_config();
    
  return result;
}
```

Basically I backup ``QTOUCH_PTC->BURSTMODE.reg`` and restore its value after ```startPtcAcquire ```. It seems to work. I have no idea what I'm doing and I didn't find any document about the BURSTMODE register.

ladyada, 2020-07-31:
there's no documentation - we reverse engineered it from decompiling, so not surprising theres bugs :) can you please submit a PR?

closed 2020-07-31 by DeqingSun