adafruit/Adafruit_CircuitPlayground #57: Package discription is no longer accurate.

easytarget, 2020-08-17:
Library Description is incorrect. From `library.properties`:
```
sentence=All in one library to control Adafruit's Circuit Playground board.
paragraph=All in one library to control Adafruit's Circuit Playground board.  Requires no other dependencies and exposes all Circuit Playground components in a simple to use class.

```
commit: 9b216be8f76d30a5aeb5f6d2db4e096b1be00742 added a `depends=` line to `library.properties`, creating six dependencies, which in turn bring in an even larger tree of sub-dependencies.

The description is no longer accurate and should be changed to reflect this.

ladyada, 2020-08-18:
technically, those libraries are only used for demos and the library works without em! we can update that line

@siddacious wanna remove it?

easytarget, 2020-08-20:
It was just an observation on my part, the friggin library mangler in the Arduino IDE is so darn slow that I've spent more time than is good for me staring at this description, it was starting to give me a nervous tic. ;-)

siddacious, 2020-10-05:
This is fixed in the most recent release!

closed 2020-10-05 by siddacious